WangXuan95/FPGA-ftdi245fifo #11: Problem to install ftd3xx with python 3.9.13

collins4you, 2023-02-20:
Hi,
I tried to use python with my ft601 chip (an UMFT601X-B board connected with a kintex board KC701).
I've just installed the last version of anaconda (V2.3.1 with python 3.9.13). Unfortunately when i try to install the FT60X Python Software exemple with this command : "python setup.py install" I get this answer :

_(base) C:\Users\xxxx\Downloads\D3XXPython_Release1.0\D3XXPython_Release1.0>python setup.py install
Traceback (most recent call last):
  File "C:\Users\xxxxx\Downloads\D3XXPython_Release1.0\D3XXPython_Release1.0\setup.py", line 7, in <module>
    from distutils.command.build_py import build_py_2to3 as build_py
ImportError: cannot import name 'build_py_2to3' from 'distutils.command.build_py' (C:\Users\xxxxx\Anaconda3\lib\site-packages\setuptools\_distutils\command\build_py.py)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\xxxxx\Downloads\D3XXPython_Release1.0\D3XXPython_Release1.0\setup.py", line 10, in <module>
    raise ImportError("build_py_2to3 not found in distutils - it is required for Python 3.x")
ImportError: build_py_2to3 not found in distutils - it is required for Python 3.x_

It seems there is a problem with the new version of setuptools (v65.6.3 in anaconda) :
"setuptools v58.0.0 has a breaking change, first included in Python 3.10, where build_py_2to3 was removed" 
(read there : https://stackoverflow.com/questions/75085575/importerror-cannot-import-name-build-py-2to3-from-distutils-command-build-py)
The python driver of FTDI probably use old libraries and become obsolete now. So, if you have a simple solution, why not...
Bytheway i'll have to explore the python wrapping of ftdi driver...

collins4you, 2023-02-28:
Finelly i drop all the ftd3xx files and driver dll in the python directory, it works
Just need to replace in file USB_FTX232H_FT60X.py : 
usb = ftd3xx.create(device_id, ftd3xx._ftd3xx_win32.FT_OPEN_BY_INDEX)
usb = ftd3xx.create(device_id, ftd3xx._ftd3xx_linux.FT_OPEN_BY_INDEX)

WITH : 
usb = ftd3xx.create(device_id, ftd3xx._ft.FT_OPEN_BY_INDEX)
usb = ftd3xx.create(device_id, ftd3xx._ft.FT_OPEN_BY_INDEX)
[image]

WangXuan95, 2023-04-09:
Thanks a lot for your solution! I'll try it and update the repo later.

WangXuan95, 2023-06-03:
Hello,

Now it supports to install `ftd3xx` library using `pip`:
`python -m pip install ftd3xx`

I've updated corresponding README document.

closed 2023-06-03 by WangXuan95