WangXuan95/FPGA-ftdi245fifo #13: Implemented design on a Zynq no PC -> FPGA communication

gswdh, 2023-10-24:
I have this core implemented in a Zynq with a DMA implemented to send and receive data from the FTDI core which is connected to an FT601. However, sending data from the PC to the FPGA does not work, FPGA -> PC does work.

I have tested my DMA on the Zynq via a loop back - working great.

A loop back with or without a FIFO on the IP core does not work.

With the DMA connected to the IP core does not work.

I did have to manually implement two of the parameters as follows due to synth error in Vivado.

```
localparam  CHIP_EW               = 2;
localparam  CHIP_DRIVE_AT_NEGEDGE = 0;
```

The project I have it implemented in is in this project https://github.com/gswdh/CameraFPGA/tree/develop

I wonder if there's any advice or example project of implementing this core on a Xilinx device.

josephorender, 2024-04-19:
I have the same issue. I am using a Zedboard with a UMFT601x FMC card. Here is the block design

[image]

[image]

[image]

Nothing ever comes out the axi stream.

gswdh, 2024-04-19:
@josephorender what happens if you send more data? Perhaps there needs to be enough to justify a flush somewhere?

josephorender, 2024-04-19:
@gswdh I pulled your project and noticed you are using Vivado 2023.2. Im using 2023.1. Either way, I think the out of context runs are causing problems with this guys code. Vivado will delete half the tristate buffers in one of his fifo blocks. You need to set this:
[image]
and it will keep the buffers. I followed the Critical warnings until they were gone, but I can still get the loopback to work even with timing currently failing. 
[image]
Hope this helps!

gswdh, 2024-04-19:
Nice work @josephorender unfortunealy I decided to drop the FT601 for the gigabit ethernet on the Zynq. Is your issue solved now?

josephorender, 2024-04-19:
It is. I bet they were related. You can probably close this one out.

gswdh, 2024-04-19:
Thanks, all the best!

closed 2024-04-19 by gswdh